Rahix/tbot #14: tbot installation

srnae05, 2020-01-15:
After installation as mentioned in documentation, tbot selftest returns tbot: command not found.
completions.sh file is added to bashrc file.

Can you please help ?
Thanks !

Rahix, 2020-01-15:
Hello @srnae05,

> I am a beginner in linux.

Not sure how much beginner, so if you know most of this already, ignore it:  On Linux (or more generally, in all POSIX systems) the shell searches for commands in the directories given in the `$PATH` environment variable.  You can take a look at this list using this command:

```shell-session
❯ echo $PATH
/usr/local/bin:/usr/bin:/bin:/usr/local/sbin
```

Now, in the installation instructions of `tbot`, you were supposed to run the install command like this:

```shell-session
❯ python3 setup.py install --user
[...]
```

Without the `--user`, this command would have installed tbot to `/usr/local/bin`, but your user does not have access rights to that directory.  So instead, using the `--user` flag, it is installed in your home-directory, more specifically into `~/.local/bin`.  This directory is not in your `$PATH` by default so you need to add it manually.

To do this, you have to add another command to your `.bashrc`:

```bash
export PATH=~/.local/bin:$PATH
```

Once you then open a new shell session you should be able to run `tbot` :)

srnae05, 2020-01-16:
Hello @Rahix 

Thank you for the reply.
My PATH variable already has the values that you had mentioned. I had already installed using the --user command. What I did not do was export PATH=~/.local/bin:$PATH. I did this now, it still does not work.

I have two observations
- There is no tbot folder or file under /usr/local/bin
- In the documentation, there is a step that say source /path/to/tbot/completions.sh to ~.bashrc. This is how I did it. In the terminal, I used the source /mypathToTbot/completions.sh. However when I open the ~.bashrc, this is not reflected. 

I believe I am doing something wrong here.

Rahix, 2020-01-16:
> In the documentation, there is a step that say source /path/to/tbot/completions.sh to ~.bashrc.
This is how I did it. In the terminal, I used the source /mypathToTbot/completions.sh. However when I open the ~.bashrc, this is not reflected.

You have to add the source command as a line in your `~/.bashrc` file:

```bash
source /mypathToTbot/completions.sh
```
> I believe I am doing something wrong here since my self test is failing

Can you please post the entire output tbot gives you?

> When I execute other .py files, i get the same error KeyError: 'hello_world'. I believe that tbot is not able to find the function definition of hello_world in a .py file. How do I fix this ?

What exact command were you running?  What was the name of the python file?

srnae05, 2020-01-17:
> ```shell
> source /mypathToTbot/completions.sh
> ```
After adding this, the self test passes. Thanks !

> What exact command were you running? What was the name of the python file?

I created a file tc.py with hello_world function in my home folder. When I execute using tbot hello_world, it fails. Here is the output
`(base) root@srnae05:/home/srnae05/ACC/testCases# tbot hello_world
Warning: Failed to load /home/srnae05/ACC/testCases/tc.py:
    Traceback (most recent call last):
      File "/root/.local/lib/python3.7/site-packages/tbot-0.8.0-py3.7.egg/tbot/loader.py", line 160, in collect_testcases
        module = load_module(f)
      File "/root/.local/lib/python3.7/site-packages/tbot-0.8.0-py3.7.egg/tbot/loader.py", line 136, in load_module
        module_spec.loader.exec_module(module)
      File "<frozen importlib._bootstrap_external>", line 724, in exec_module
      File "<frozen importlib._bootstrap_external>", line 860, in get_code
      File "<frozen importlib._bootstrap_external>", line 791, in source_to_code
      File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
      File "/home/srnae05/ACC/testCases/tc.py", line 4
        def hello_world()
                        ^
    SyntaxError: invalid syntax

tbot starting ...
├─Exception:
│   Traceback (most recent call last):
│     File "/root/.local/lib/python3.7/site-packages/tbot-0.8.0-py3.7.egg/tbot/main.py", line 295, in main
│       func = testcases[tc]
│   KeyError: 'hello_world'
├─────────────────────────────────────────
└─FAILURE (0.220s)`

Rahix, 2020-01-17:
The command-output already points to the issue.  _tbot_ complains that it can't load your testcase file:

```text
Warning: Failed to load /home/srnae05/ACC/testCases/tc.py:
```

Below that you can see the traceback of the failure, which ends with an error message:

```text
File "/home/srnae05/ACC/testCases/tc.py", line 4
   def hello_world()
   ^
   SyntaxError: invalid syntax
```

You'll have to figure out what you did wrong, but from the output I suspect you forgot the `:` after the function definition ;)

closed 2020-03-11 by Rahix