Rahix/tbot #25: Missing documentation for special character: Background

niecore, 2020-07-20:


Rahix, 2020-07-22:
Hi!

I think I've left it out of documentation because it has a few nasty side-effects.  If a program in the background still outputs to the console, this will silently break things.  Ideally, a proper `Background` special symbol would always redirect output to prevent all console clobbering.  E.g.

```python
# Automatically adds redirects to /dev/null
m.exec0("python", "-m", "http.server", linux.Background)

# Redirect stdout
m.exec0("python", "-m", "http.server", linux.Background(stdout="/tmp/http.log"))

# Redirect stdout & stderr
m.exec0("python", "-m", "http.server", linux.Background(stdout="/tmp/http.log", stderr="/tmp/http-error.log"))
```

But this, of course, needs to be implemented first ...

closed 2020-10-13 by Rahix

Rahix, 2020-10-13:
Finally got around to adding this:  https://tbot.tools/modules/machine_linux.html#tbot.machine.linux.Background

commit af189715 (2020-10-13): linux.special: Rework Background token

`linux.Background` has a number of issues related to command-output
clobbering the console.  Make this special token supress all output by
default and provide convenient file-redirection to save stdout/err.

With this change, `linux.Background` becomes much more usable, thus also
document it properly now.

Fixes #25.