Rahix/tbot #26: Board Linux shall have a way to be soft reseted (`reboot

niecore, 2020-07-24:
It would be great to the trigger a soft reset with `reboot` for some testcases.
Unfortunately `exec0 reboot` will not terminate because the prompt will not be shown again.
The `softreset` function could handle the relogin as well.

Rahix, 2020-07-24:
My current pattern for soft-resets is this:

```python
with contexlib.ExitStack() as cx:
    b = cx.enter_context(tbot.acquire_board(lh))
    lnx = cx.enter_context(tbot.acquire_linux(b))

    ...
    
    ch = lnx.open_channel("systemctl", "reboot")
    # Stuff the channel back into the board-machine ...
    b.ch = ch

    # Reinitialize the linux-machine which will wait for the reboot to
    # reach the login-prompt again.
    lnx = cx.enter_context(tbot.acquire_linux(b))

    ...
```

I know, this isn't the pretties way to do it and I was also thinking that maybe it would be useful to have a way for doing soft-resets with a single method call.  But I'm not really sure how to model this without making a lot of assumptions about the user, so we'll have to see ...

For now, this explicit re-initialization is the pattern I would suggest you use.

avylove, 2022-07-14:
I wasn't able to get the example to work. It would hang once Linux would shutdown, but wouldn't power cycle.
This does seem to work for me

```python
  re_poweroff = re.compile(b'reboot: Power down\s{0,10}')
  shutdown_cmd = ("sudo", "systemctl", "poweroff")

  with tbot.ctx.request(tbot.role.BoardLinux) as serial:
      with serial.run(*shutdown_cmd) as shutdown:
          shutdown.read_until_prompt(prompt=re_poweroff, timeout=60)
          shutdown._pre_terminate()
          shutdown._proxy_alive = False
```

I would prefer not to use private attributes, but I couldn't get it to work cleanly otherwise. I tried running `serial.ch.read_until_prompt()`, but it just hung like before.

Any advice on achieving something similar without accessing private attributes?

Rahix, 2022-07-15:
I am not 100% sure what you are trying to do...  You want to first power-off the system and then have tbot power-cycle it, is that correct?  In that case, try something like this:

```python
with tbot.ctx.request(tbot.role.BoardLinux, exclusive=True) as lnx:
    # Do something before reboot
    lnx.exec0("uname", "-a")

    # Power-down the system
    ch = lnx.open_channel("systemctl", "poweroff")
    ev = tbot.log.EventIO(["powerdown"], "System powering down...")
    with ch.with_stream(ev):
        ev.prefix = "   <> "
        ch.expect(["reboot: Power down", "reboot: System halted"])

with tbot.ctx.request(tbot.role.BoardLinux, reset=True) as lnx:
    # Do something after reboot
    lnx.exec0("uname", "-a")
```

FWIW, this issue was originally about performing a soft-reset _without_ power-cycling the system.  For that, things work a bit differently...

avylove, 2022-07-21:
Thanks! Using `Channel.expect()` is much cleaner.

For my use case, either a reboot or a shutdown will work. I just need the OS to shutdown cleanly and not just power cycle. The issue I had with the reboot is it just hung. Perhaps it also needs `Channel.expect()`?

avylove, 2022-08-12:
Turns out I do need to reboot. I tried to implement what you have in https://github.com/Rahix/tbot/issues/26#issuecomment-663580867. It works if I'm just running additional commands on the serial console within that same context block, but, if I call other code after this that requests the `LinuxBoard` role and uses the channel, I'll get a `ChannelTakenException`. I'm guessing the global context is sharing the original machine object and not the second one?

franzflasch, 2023-07-07:
I would also need to do a reboot without a power cylcle in between. How can this be done cleanly?

kvmajo, 2026-08-19:
We've hit the same need (reboot without a power-cycle) and ended up with a variant of the pattern @Rahix posted above, wrapped in a context manager so it can be reused across tests:

```python
import typing
from contextlib import contextmanager
from tbot.machine import board, channel

@contextmanager
def reinitiated_machine(lnx: board.LinuxBootLogin, ch: typing.Optional[channel.Channel] = None):
    """
    Reinitialize a LinuxBootLogin machine's shell/state in place, without a
    full teardown (and therefore without power-cycling the board).

    Use this after the test itself has triggered a reboot (e.g. via
    `lnx.exec0("systemctl", "reboot")`), to reconnect and wait for the
    login prompt again on the same channel.

    An optional `ch` can be passed if the channel changed (e.g. a new
    channel was opened for the reboot command itself); otherwise the
    original machine's channel is reused.
    """
    if not isinstance(lnx, board.LinuxBootLogin):
        raise TypeError("Machine does not inherit LinuxBootLogin")

    # Reset boot-detection state so `_init_machine()`/`init()` wait for a
    # fresh login prompt instead of assuming the machine is already up.
    lnx._linux_init_event = None
    lnx._boot_start = None

    if ch is not None:
        lnx.ch = ch

    with lnx._init_machine(), lnx._init_shell():
        lnx.init()
        yield lnx
```

Usage:

```python
with tbot.ctx() as cx:
    with cx.request(tbot.role.BoardLinux) as lnx:
        lnx.exec0("systemctl", "reboot")
        with reinitiated_machine(lnx) as lnx:
            lnx.exec0("echo", "Command run after reboot")
```

Note: if you pass a new `ch`, you'll likely want to re-register any death-strings on it that were set on the original channel (e.g. via `ch.add_death_string(...)`) — they don't carry over automatically.

It works, but it's built entirely on private `LinuxBootLogin` internals (`_linux_init_event`, `_boot_start`, `_init_machine()`, `_init_shell()`), so it's fragile across tbot versions. Sharing in case it helps others here, or as a data point for what a built-in soft-reset API would need to do internally.