Rahix/tbot #28: tbot fails (circular import)

zkrx, 2020-10-08:
On my system (Arch Linux), `tbot selftest` (or any other tbot invocation) fails with the following error:

`[laptop tbot]$ ~/.local/bin/tbot selftest
Traceback (most recent call last):
  File "/home/laptop/.local/bin/tbot", line 8, in <module>
    from tbot.main import main
  File "/home/laptop/.local/lib/python3.8/site-packages/tbot-0.8.3-py3.8.egg/tbot/__init__.py", line 22, in <module>
    from . import selectable
  File "/home/laptop/.local/lib/python3.8/site-packages/tbot-0.8.3-py3.8.egg/tbot/selectable.py", line 18, in <module>
    from tbot.machine import connector, linux, board
  File "/home/laptop/.local/lib/python3.8/site-packages/tbot-0.8.3-py3.8.egg/tbot/machine/__init__.py", line 1, in <module>
    from . import board
  File "/home/laptop/.local/lib/python3.8/site-packages/tbot-0.8.3-py3.8.egg/tbot/machine/board/__init__.py", line 4, in <module>
    from .uboot import UBootShell, UBootAutobootIntercept
  File "/home/laptop/.local/lib/python3.8/site-packages/tbot-0.8.3-py3.8.egg/tbot/machine/board/uboot.py", line 24, in <module>
    from ..linux import special
  File "/home/laptop/.local/lib/python3.8/site-packages/tbot-0.8.3-py3.8.egg/tbot/machine/linux/__init__.py", line 34, in <module>
    from . import build
  File "/home/laptop/.local/lib/python3.8/site-packages/tbot-0.8.3-py3.8.egg/tbot/machine/linux/build/__init__.py", line 1, in <module>
    from .machine import BuildMachine
  File "/home/laptop/.local/lib/python3.8/site-packages/tbot-0.8.3-py3.8.egg/tbot/machine/linux/build/machine.py", line 20, in <module>
    from . import toolchain
  File "/home/laptop/.local/lib/python3.8/site-packages/tbot-0.8.3-py3.8.egg/tbot/machine/linux/build/toolchain.py", line 22, in <module>
    class Toolchain(abc.ABC):
  File "/home/laptop/.local/lib/python3.8/site-packages/tbot-0.8.3-py3.8.egg/tbot/machine/linux/build/toolchain.py", line 26, in Toolchain
    def enable(self, host: linux.LinuxMachine) -> None:
AttributeError: partially initialized module 'tbot.machine.linux' has no attribute 'LinuxMachine' (most likely due to a circular import)`

The issue appeared between 0.7.1 (which works) and 0.8.0 (which doesn't work). Latest 0.8.3 doesn't work either.

`
[laptop tbot]$ python --version
Python 3.8.5
`

I installed using: `python3 setup.py install --user`

Am I missing something :)?

Rahix, 2020-10-08:
Hmm, currently on a Fedora, also with Python 3.8.5.  I can't reproduce unfortunately ... Does a development installation with

```bash
python3 setup.py develop --user
```

work, maybe?  I'll try to set up an Arch box, hopefully that lets me reproduce it ...

Rahix, 2020-10-08:
Okay, on a minimal ArchLinux, where I just installed `pacman -S python python-pip git` and then installed tbot with

```bash
python3 setup.py install --user
```

I can't reproduce it ... So there must be something differing in your setup from this ...

zkrx, 2020-10-08:
`python3 setup.py develop --user` on 0.8.3 solves it. Running `python3 setup.py install --user` again makes the problem reappear.

zkrx, 2020-10-08:
I've tried deleting ~/.local/ and running `python3 setup.py install --user` again, the problem persists.

Rahix, 2020-10-08:
Weird ... I'll have to take a closer look.  For now, I hope you can work with the development installation :)

(The difference is that 'install' copies the module to `site-packages` while 'develop' only places a symlink to your repo-checkout there ...)

zkrx, 2020-10-08:
Yep, that should do it. Perhaps the issue is on my end. Thanks for looking into this, don't hesitate if you need something.

zkrx, 2020-10-14:
A colleague experiences the same issue on his machine with Fedora 32 and Python 3.8.3. `python3 setup.py develop --user` solves it there too.

Rahix, 2020-10-15:
Wait ... This stack-frame does not make any sense:

```text
  File "/home/laptop/.local/lib/python3.8/site-packages/tbot-0.8.3-py3.8.egg/tbot/machine/linux/build/toolchain.py", line 26, in Toolchain
    def enable(self, host: linux.LinuxMachine) -> None:
```

There is **no** `tbot/machine/linux/build/toolchain.py` in any current version of tbot.  This file was removed a long time ago ... Why is it showing up here?

I think I got an idea ...

```shell-session
$ git clone https://github.com/Rahix/tbot.git
$ cd tbot
$ git switch --de v0.7.1
$ python setup.py install --user
$ git switch master
$ python setup.py install --user
$ tbot selftest
[...]
AttributeError: partially initialized module 'tbot.machine.linux' has no attribute 'LinuxMachine' (most likely due to a circular import)
```

I suppose you and your colleague had installed a `<0.8` version in the past and now updated to `>=0.8`?  This sounds familiar, I had a report like this in the past ... Can you try the steps from [_Installation successful, running tbot always throws an exception_](https://tbot.tools/installation.html#installation-successful-running-tbot-always-throws-an-exception)?  In the meantime I'll try to find out why the hell setuptools breaks the installation like that ...

zkrx, 2020-10-15:
> I suppose you and your colleague had installed a <0.8 version in the past and now updated to >=0.8?

Yes, exactly. We were on 0.7.1 and I'm only upgrading now.

> Can you try the steps from Installation successful, running tbot always throws an exception? 

Oof sorry for missing this... This works, thanks a lot.

Rahix, 2020-10-15:
No problem, glad we could solve it!  Still don't understand why this is failing, though ... Surely setuptools can deal with files being removed from a package on update.  Have to investigate some more ...

closed 2022-05-13 by Rahix