Rahix/tbot #30: uboot_testpy: POWERON and POWEROFF in the wrong sequence

xypron, 2020-10-09:
I have created scripts to manage my test environment. They are available at
https://github.com/xypron/tbotrunner/tree/orangepipc

Target **interactive_uboot** runs as expected.

Target **uboot_testpy** powers the board on before collecting the unit tests and powers it off before executing the unit tests. Obviously this leads to failures in all tests.

A file 'test/py/u_boot_boardenv_tbot_OrangePi PC.py' is created. But that file does not match the name of the defconfig. Why should the name property of the Board class be used here?

Files u-boot-test-console  u-boot-test-flash  u-boot-test-quit  u-boot-test-reset are missing.

```
user@rpi2:~/workspace/tbotrunner$ tbot -l labconfig.py -b boardconfig.py -vv uboot_testpy
tbot starting ...
├─Calling uboot_testpy ...
│   ├─Message: MyLabHost.build
│   ├─[rpi2] bash --norc --noprofile
│   ├─Calling uboot_setup_testhooks ...
│   │   ├─[rpi2] mkdir -p /tmp/tbot-workdir
│   │   ├─[rpi2] test -d /tmp/tbot-workdir/uboot-testpy-tbot
│   │   ├─Creating FIFOs ...
│   │   ├─[rpi2] rm -rf /tmp/tbot-workdir/uboot-testpy-tbot/fifo_console_send
│   │   ├─[rpi2] mkfifo /tmp/tbot-workdir/uboot-testpy-tbot/fifo_console_send
│   │   ├─[rpi2] rm -rf /tmp/tbot-workdir/uboot-testpy-tbot/fifo_console_recv
│   │   ├─[rpi2] mkfifo /tmp/tbot-workdir/uboot-testpy-tbot/fifo_console_recv
│   │   ├─[rpi2] rm -rf /tmp/tbot-workdir/uboot-testpy-tbot/fifo_commands
│   │   ├─[rpi2] mkfifo /tmp/tbot-workdir/uboot-testpy-tbot/fifo_commands
│   │   ├─[rpi2] cat /tmp/tbot-workdir/uboot-testpy-tbot/tbot-scripts.sha256
│   │   │    ## 2d30892b61eb713ce9413e06c4f2a0cd00d2a74b6b8c2ac6624e1e49909b1897
│   │   ├─Hooks are up to date, skipping deployment ...
│   │   ├─Adding hooks to $PATH ...
│   │   ├─[rpi2] echo " ${PATH}"
│   │   │    ##  /home/user/.local/bin:/home/user/.local/bin:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games
│   │   ├─[rpi2] export PATH=/tmp/tbot-workdir/uboot-testpy-tbot:/home/user/.local/bin:/home/user/.local/bin:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games
│   │   ├─Open console & command channels ...
│   │   ├─[rpi2] /tmp/tbot-workdir/uboot-testpy-tbot/tbot-console
│   │   ├─[rpi2] /tmp/tbot-workdir/uboot-testpy-tbot/tbot-commands
│   │   └─Done. (0.111s)
│   ├─Calling uboot_checkout ...
│   │   ├─Builder: orangepipc
│   │   ├─[rpi2] test -d /tmp/tbot-workdir/uboot-orangepipc/.git
│   │   ├─[rpi2] git -C /tmp/tbot-workdir/uboot-orangepipc fetch
│   │   │    ## remote: Enumerating objects: 1, done.
│   │   │    ## remote: Counting objects: 100% (1/1), done.
│   │   │    ## remote: Total 1 (delta 0), reused 0 (delta 0), pack-reused 0
│   │   │    ## Unpacking objects: 100% (1/1), done.
│   │   │    ## From https://gitlab.denx.de/u-boot/u-boot
│   │   │    ##    0570938e3c..0437cc4155  master     -> origin/master
│   │   └─Done. (1.822s)
│   ├─[rpi2] test -e /tmp/tbot-workdir/uboot-orangepipc/.config
│   ├─[rpi2] test -e /tmp/tbot-workdir/uboot-orangepipc/include/autoconf.mk
│   ├─[rpi2] picocom -b 115200 /dev/serial/by-path/platform-3f980000.usb-usb-0:1.1.3:1.0-port0
│   ├─POWERON (OrangePi PC)
│   ├─[rpi2] relay-card off
│   ├─[rpi2] sd-mux-ctrl -v 0 -td
│   ├─[rpi2] relay-card on
│   ├─UBOOT (my-board-u-boot)
│   │    <> picocom v3.1
│   │    <> 
│   │    <> port is        : /dev/serial/by-path/platform-3f980000.usb-usb-0:1.1.3:1.0-port0
│   │    <> flowcontrol    : none
│   │    <> baudrate is    : 115200
│   │    <> parity is      : none
│   │    <> databits are   : 8
│   │    <> stopbits are   : 1
│   │    <> escape is      : C-a
│   │    <> local echo is  : no
│   │    <> noinit is      : no
│   │    <> noreset is     : no
│   │    <> hangup is      : no
│   │    <> nolock is      : no
│   │    <> send_cmd is    : sz -vv
│   │    <> receive_cmd is : rz -vv -E
│   │    <> imap is        : 
│   │    <> omap is        : 
│   │    <> emap is        : crcrlf,delbs,
│   │    <> logfile is     : none
│   │    <> initstring     : none
│   │    <> exit_after is  : not set
│   │    <> exit is        : no
│   │    <> 
│   │    <> Type [C-a] [C-h] to see available commands
│   │    <> Terminal ready
│   │    <> 
│   │    <> U-Boot SPL 2020.10-00607-g0570938e3c (Oct 09 2020 - 19:50:13 +0000)
│   │    <> DRAM: 1024 MiB
│   │    <> Trying to boot from MMC1
│   │    <> 
│   │    <> 
│   │    <> U-Boot 2020.10-00607-g0570938e3c (Oct 09 2020 - 19:50:13 +0000) Allwinner Technology
│   │    <> 
│   │    <> CPU:   Allwinner H3 (SUN8I 1680)
│   │    <> Model: Xunlong Orange Pi PC
│   │    <> DRAM:  1 GiB
│   │    <> MMC:   mmc@1c0f000: 0
│   │    <> Loading Environment from FAT... *** Warning - bad CRC, using default environment
│   │    <> 
│   │    <> In:    serial
│   │    <> Out:   serial
│   │    <> Err:   serial
│   │    <> Net:   phy interface0
│   │    <> eth0: ethernet@1c30000
│   │    <> starting USB...
│   │    <> Bus usb@1c1a000: USB EHCI 1.00
│   │    <> Bus usb@1c1a400: USB OHCI 1.0
│   │    <> Bus usb@1c1b000: USB EHCI 1.00
│   │    <> Bus usb@1c1b400: USB OHCI 1.0
│   │    <> Bus usb@1c1c000: USB EHCI 1.00
│   │    <> Bus usb@1c1c400: USB OHCI 1.0
│   │    <> Bus usb@1c1d000: USB EHCI 1.00
│   │    <> Bus usb@1c1d400: USB OHCI 1.0
│   │    <> scanning bus usb@1c1a000 for devices... 1 USB Device(s) found
│   │    <> scanning bus usb@1c1a400 for devices... 1 USB Device(s) found
│   │    <> scanning bus usb@1c1b000 for devices... 1 USB Device(s) found
│   │    <> scanning bus usb@1c1b400 for devices... 1 USB Device(s) found
│   │    <> scanning bus usb@1c1c000 for devices... 1 USB Device(s) found
│   │    <> scanning bus usb@1c1c400 for devices... 1 USB Device(s) found
│   │    <> scanning bus usb@1c1d000 for devices... 1 USB Device(s) found
│   │    <> scanning bus usb@1c1d400 for devices... 1 USB Device(s) found
│   │    <>        scanning usb for storage devices... 0 Storage Device(s) found
│   │    <> Hit any key to stop autoboot:  0 
│   │    <> => 
│   ├─[rpi2] tee '/tmp/tbot-workdir/uboot-orangepipc/test/py/u_boot_boardenv_tbot_OrangePi PC.py' >/dev/null
│   ├─[rpi2] cd /tmp/tbot-workdir/uboot-orangepipc
│   ├─[rpi2] ./test/py/test.py --build-dir . --board-type 'tbot-OrangePi PC'
│   │    ## +u-boot-test-flash tbot-OrangePi PC na
│   │    ## ========================================================================= test session starts =========================================================================
│   │    ## platform linux -- Python 3.7.3, pytest-3.10.1, py-1.7.0, pluggy-0.8.0
│   │    ## rootdir: /tmp/tbot-workdir/uboot-orangepipc/test/py, inifile: pytest.ini
│   │    ## collected 327 items                                                                                                                                                   
│   ├─[rpi2] relay-card off
│   ├─POWEROFF (OrangePi PC)
│   ├─[rpi2] relay-card off
│   │    ## 
│   │    ## test/py/tests/test_000_version.py E                                                                                                                             [  0%]
│   │    ## test/py/tests/test_bind.py ss                                                                                                                                   [  0%]
│   │    ## test/py/tests/test_button.py s                                                                                                                                  [  0%]
│   │    ## test/py/tests/test_dfu.py s                                                                                                                                     [  0%]
│   │    ## test/py/tests/test_dm.py EE
```

Rahix, 2020-10-10:
Hi,

first of all, I think a space in your board name is not something I properly care for [here](https://github.com/Rahix/tbot/blob/master/tbot/tc/uboot/testpy.py#L272-L274).  Not sure what problems this will cause (if any), but to be safe I'd replace that with a `'-'`.

> Target uboot_testpy powers the board on before collecting the unit tests and powers it off before executing the unit tests. Obviously this leads to failures in all tests.

Powering on before starting test/py is expected.  Then, after test/py has collected all testcases, it will issue a board reset.  That's why the board is powered off again at the beginning of tests.  If everything works, tbot will first call your `poweroff()` method and then `poweron()`.  What is weird is that only the first command from your `poweron()` method is sent.

Digging into your lab-config, I see

```python
    def build(self):
        tbot.log.message(tbot.log.c("Message").yellow.bold + ": MyLabHost.build")
        return self
```

Try making that a

```python
    def build(self):
        tbot.log.message(tbot.log.c("Message").yellow.bold + ": MyLabHost.build")
        return self.clone()
```

The problem here is that `uboot_testpy` expects to have separate connections to lab-host and build-host but you gave it the same for both.  So it's trying to execute the power commands on the same session where test/py is currently running!  I thought, I had built in checks to catch that but apparently something slipped through ... I'll see how I can protect against this mistake.

> Files u-boot-test-console u-boot-test-flash u-boot-test-quit u-boot-test-reset are missing.

Those are in `/tmp/tbot-workdir/uboot-testpy-tbot` if you're curious.

Rahix, 2020-10-10:
Pushed commit 39f21f086030 ("test/py: Issue a warning for incorrectly configured lab-hosts") which will warn about and attempt to fix such configurations.

xypron, 2020-10-10:
@Rahix 
Thank you for reviewing my code. With the suggested changes is works as expected. 

Adding the following information to the online documentation would have helped me:
* description of the build() and enable() methods of a Lab instance detailing how they work
* how to setup a a build host that is not the lab host
* how to use DistroToolchain(). I never got it to use the cross-compiler for the board.

Best regards

Heinrich

Rahix, 2020-10-11:
Thanks for the feedback, I really appreciate it!

> description of the build() and enable() methods of a Lab instance detailing how they work

Yeah, the docs about this could really use some touch up, I agree.  I've been holding off from that because I don't like the current implementation (especially of `build()`) very much and I am in the process of overhauling that to make it easier and more flexible.

> how to setup a a build host that is not the lab host

Wow, the docs actually don't show any example of this ... I'll add one but in the mean time, I hope this helps:

```python
class SomeBuildHost(connector.SSHConnector, linux.Bash, linux.Builder):
    hostname = "some-build-host.example.com"

    @property
    def toolchains(self):
        ...

class SomeLabHost(..., linux.Lab):
    def build(self):
        SomeBuildHost(self)
```

and later, you can copy files back to the labhost in a testcase like this:

```python
from tbot.tc import shell

...

target = lh.workdir / "some-file"
with lh.build() as bh:
    source = bh.workdir / "some-file"

    shell.copy(source, target)
```

> how to use DistroToolchain(). I never got it to use the cross-compiler for the board.

Hmm, doing this in your labconfig doesn't work?

```python
    @property
    def toolchains(self) -> typing.Dict[str, linux.build.Toolchain]:
        """ Define Toolchains
        """
        return {
            "arm": linux.build.DistroToolchain("arm", "arm-linux-gnueabihf-"),
            "arm64": linux.build.DistroToolchain("arm64", "aarch64-linux-gnu-"),
            "riscv": linux.build.DistroToolchain("riscv", "riscv64-linux-gnu-"),
        }
```

xypron, 2020-10-11:
@Rahix 
I just rechecked DistroToolchain. It is usable. Probably in my early experiments I had something wrong in introducing the toolchain.

closed 2020-10-11 by xypron