Rahix/tbot #31: kconfig and relative working directory path

xypron, 2020-10-11:
The working directory of the lab  may be defined as a relative path:

```
def workdir(self):
    linux.Workdir.static(self, f"tbot-workdir")
```
This leads to a failure in kconfig_set_enabled:

```
│   ├─Calling kconfig_set_enabled ...
│   │   ├─Enabling CONFIG_CMD_BOOTEFI_SELFTEST option ...
│   │   ├─[rpi2] sed -i '/^\(# \)\?CONFIG_CMD_BOOTEFI_SELFTEST\(=[ym]\| is not set\)$/cCONFIG_CMD_BOOTEFI_SELFTEST=y' tbot-workdir/uboot-orangepipc/.config
│   │   │    ## sed: can't read tbot-workdir/uboot-orangepipc/.config: No such file or directory
│   │   └─Fail. (0.014s)
│   └─Fail. (36.598s)
```
sed is executed in the work directory. So the work directory path must not be prepended.

Rahix, 2020-10-11:
Hm, a relative path for the working directory isn't an intentional feature to be honest ... I actually think this should be explicitly disallowed because it breaks a lot of internal assumptions.  If you do want a feature for this, I'd say it would make sense to introduce a new `Workdir` type that only works on the current host (i.e. not on remote machines) and is always relative to the tbot CWD.

xypron, 2020-10-11:
So I convert this relative path to an absolute one:

```
    @property
    def workdir(self):
        mypath = os.path.dirname(os.path.realpath(__file__)) + '/tbot-workdir'
        return linux.Workdir.static(self, mypath)
```

Rahix, 2020-10-12:
Yeah, that's a workaroud I'm also using in some of my own configs.  Though as stated, I think it would be best to implement a proper "relative workdir" feature for this at some point.  Let's repurpose this issue to keep track of that.