Rahix/tbot #33: kconfig_set_enabled misses to call olddefconfig

xypron, 2020-10-11:
kconfig.enable can be used to enable a Kconfig option. But this is not enough. One has to run 'make olddefconfig' to avoid requests for input during the uboot_build target:

```
│   ├─Calling kconfig_set_enabled ...
│   │   ├─Enabling CONFIG_UNIT_TEST option ...
│   │   ├─[rpi2] sed -i '/^\(# \)\?CONFIG_UNIT_TEST\(=[ym]\| is not set\)$/cCONFIG_UNIT_TEST=y' /mnt/iscsi/user/workspace/tbotrunner/tbot-workdir/uboot-orangepipc/.config
│   │   └─Done. (0.019s)
│   ├─Calling uboot_make ...
│   │   ├─[rpi2] nproc --all
│   │   │    ## 4
│   │   ├─[rpi2] make -j 4 all
│   │   │    ## scripts/kconfig/conf  --syncconfig Kconfig
│   │   │    ## *
│   │   │    ## * Restart config...
│   │   │    ## *
│   │   │    ## *
│   │   │    ## * Unit tests
│   │   │    ## *
│   │   │    ## Unit tests (UNIT_TEST) [Y/n/?] y
│   │   │    ##   Unit tests for library functions (UT_LIB) [Y/n/?] (NEW) 
```

A good solution would be to call 'make olddefconfig' in UBootBuilder._build() after do_configure() by default.

Rahix, 2020-10-11:
Yeah, good point! I also noticed that a while back but didn't get around to fixing it ...

I think your proposed solutions is a good workaround for now but ideally this should happen as part of the kconfig testcases, I'd say?  Though in Linux, this needs toolchain environment (especially `$ARCH`) configured to behave correctly, which makes this a bit more difficult ...

Rahix, 2020-10-13:
I've pushed the suggested workaround to master but let's keep this issue open as this seems like a general problem with the kconfig testcases at this point.