Rahix/tbot #60: Paramiko SSHException: not found in known_hosts

drewwestrick, 2021-09-01:
I'm having some trouble connecting to a Raspberry Pi as a Lab config. I'm getting the follow error when I try to run the interactive_lab command. I can connect using the command ssh 172.16.100.119 without a password just fine. This connection is also in the known_hosts file.

`paramiko.ssh_exception.SSHException: Server '172.16.100.119' not found in known_hosts`

ssh_config
```
Host 172.16.100.119
  HostName 172.16.100.119
  IdentitiesOnly No
  User pi
  IdentityFile /home/drewwestrick/.ssh/rasp_pi
```

tc.py
```python
import tbot
from tbot.machine import connector, linux

class AwesomeLab(
    connector.ParamikoConnector,
    linux.Bash,
    linux.Lab,
):
    name = "172.16.100.119"
    hostname = "172.16.100.119"
    username = "pi"

    @property
    def workdir(self):
        return linux.Workdir.athome(self, "tbot-workdir")

# tbot will check for `LAB`, don't forget to set it!
LAB = AwesomeLab
```c

drewwestrick, 2021-09-01:
I think I found one workaround for now. If I add a `ignore_hostkey = True` after the username property, it allows me to make a connection. I'm still curious if paramiko uses the ~/.ssh/known_hosts file or something different to verify the server connection.

Rahix, 2021-09-03:
Hi, thanks for the report.  This looks odd indeed, I never had a problem like this... But it certainly looks like an upstream paramiko issue, I think.

To be quite honest, I would suggest that you switch to using the `connector.SSHConnector` instead.  It uses the `ssh` command exactly like you would and thus sidesteps all of those odd issues from differences between OpenSSH and Paramiko...

For performance, I very much suggest also enabling the [`use_multiplexing = True`](https://tbot.tools/modules/machine_connector.html#tbot.machine.connector.SSHConnector.use_multiplexing) option.