Rahix/tbot #65: How to access Board from BoardUboot and BoardLinux?

jneuhauser, 2022-01-27:
Hello,

I have a configuration like below where I want to access properties of my board configuration from my Uboot or Linux configuration. See the method `init()` in the class `MyUboot`. The configuration below works because `board.Connector` provides `_board` but by convention we should not rely on properties with underscore prefix.
```python
from tbot.machine import board, channel, connector

class MyBoard(
    connector.ConsoleConnector,
    board.PowerControl,
    board.Board,
):
    device_id = 0
    device_ip = "10.20.30.43"
    netmask = "255.255.255.0"
    gateway_ip = "10.20.30.254"
    relay_ip = "10.20.30.41"
	
    def connect(self, mach) -> channel.Channel:
        return mach.open_channel("picocom", "-b", "115200", "/dev/ttyUSB0")
	
    def poweron(self) -> None:
        self.host.exec0("curl", "http://{self.relay_ip}/io.cgi?relay={device_id}?state=1")
	
    def poweroff(self) -> None:
        self.host.exec0("curl", "http://{self.relay_ip}/io.cgi?relay={device_id}?state=0")

class MyUboot(
    board.Connector,
    board.UBootAutobootIntercept,
    board.UBootShell,
):
    def init(self) -> None:
        self.env("ipaddr", self._board.device_ip)
        self.env("netmask", self._board.netmask)
        self.env("gatewayip", self._board.gateway_ip)
```

Is there an intended way of accessing these objects like the context API for tests or do we need to introduce something new?

Thanks in advance, Johann

Rahix, 2022-02-10:
Okay, I've merged #66 anyway but kept this issue open so you can still report whether this is enough to accomodate your usecase.  If not, I'm happy to revise the changes again.

jneuhauser, 2022-02-11:
Sorry for the late reply, I was very busy with other things....

I took a look at the implementation, which looks very clean and is exactly what I'm looking for.
I haven't tested the implementation yet unfortunately, but will do so next week and then close the issue.

Thanks for your great work, I like tbot a lot and hopefully I can contribute some time in the future.

jneuhauser, 2022-02-22:
Works like a charm. Thank you.

closed 2022-02-22 by jneuhauser