Rahix/tbot #71: At least Python 3.8 is required by tbot/newbot

Mike8, 2022-08-04:
Running `tbot` or `newbot` always aborts with following exception:

    $ newbot
    Traceback (most recent call last):
      File "/home/mike/.local/bin/newbot", line 8, in <module>
        sys.exit(main())
      File "/home/mike/.local/lib/python3.6/site-packages/tbot/newbot.py", line 186, in main
        parser = build_parser()
      File "/home/mike/.local/lib/python3.6/site-packages/tbot/newbot.py", line 148, in build_parser
        parser.add_argument("-c", "--config", nargs=1, action="extend", default=[])
      File "/usr/lib/python3.6/argparse.py", line 1346, in add_argument
        raise ValueError('unknown action "%s"' % (action_class,))
    ValueError: unknown action "extend"

The reason for this is that `action="extend"` is used from argparse.py. This action has been introduced in Python 3.8 according to the [documentation](https://docs.python.org/3.8/library/argparse.html#action). tbot [documentation](https://tbot.tools/installation.html) say the minimum required Python version is 3.6, however.

This should be fixed either by still supporting Python 3.6 and therefore change the implementation, or, at least and much easier, by adopting the docu.

Rahix, 2022-08-04:
You're right, the documentation should be fixed in any case:  3.6 is long EOL, I had already switched CI to 3.7 but apparently not the docs.

But that aside, I wonder why this is `"extend"` in the first place.  I think `"append"` should work just as well here?

Rahix, 2022-08-04:
I guess this also shows that we are missing "frontend tests" which actually run `newbot` to check that the CLI works as expected.

Rahix, 2022-08-04:
Okay, pushed both changes.  I'll close this issue but please let me know if there are further problems I missed.  In that case I will reopen it.

closed 2022-08-04 by Rahix