Rahix/tbot #105: Timeout error while trying to reach the login prompt of the serial port over Paramiko Labhost

Adnan-Elhammoudi, 2023-12-18:
It appears that there is an error while trying to establish a connection to a serial port using Paramiko through tbot. The issue arises when encountering a Paramiko timeout reading while trying to connect to the login prompt. The code is designed to set up connections to a LabHost, a board, and a board's Linux environment. However, it fails to reach the expected login prompt within the specified timeout

Lab conf:

`import tbot
import contextlib
import json
from tbot.machine import connector, board, linux
from tbot.machine.linux.linux_shell import LinuxShell

class StartBashInitializer(tbot.machine.Initializer):
    @contextlib.contextmanager
    def _init_machine(self):
        linux.util.wait_for_shell(self.ch)
        self.ch.sendline("bash --norc --noprofile")
        yield None

class RemoteHost(
    connector.ParamikoConnector,
    StartBashInitializer,
    linux.Bash,
):
    with open('config/config.json', 'r') as file:
        data = json.load(file)
        hostname = data['LabHost']['ip_address']
        username = data['LabHost']['username']
        authenticator =linux.auth.PasswordAuthenticator(data['LabHost']['password'])

class Device(connector.ConsoleConnector,
board.Board):

        with open('config/config.json', 'r') as file:
            data = json.load(file)
            baudrate = data['DUT']['baudrate']
            serial_port  = data['DUT']['serial_port']
            file.close()
            
        tbot.log.message("Waiting for serial logging...") 
        def connect(self, mach):
            ch = mach.open_channel("picocom", "-b", str(Device.baudrate), Device.serial_port)
            ch.slow_send_delay = 0.01
            ch.slow_send_chunksize = 32
            ch.sendcontrol("M")
            return ch        
        # def poweron(self):
        #     self.ch.sendline("echo Hi!")
            
        # def poweroff(self):
        #     pass

class DeviceLinux(
board.Connector, 
board.LinuxBootLogin, 
linux.Ash
):  

    username = "root"
    password = None
    # password_prompt=None
    # askfirst_prompt= 'Please press Enter to activate this console.'
    # login_prompt =  tbot.Re(r"~ # .{0,100}")
    login_prompt = "~#"
    # login_delay=5
    # no_password_timeout = None
    boot_timeout =5

def register_machines(ctx):
    ctx.register(RemoteHost, tbot.role.LabHost)
    ctx.register(DeviceLinux, tbot.role.BoardLinux)
    ctx.register(Device, tbot.role.Board)`

Adnan-Elhammoudi, 2023-12-18:
@Rahix

Rahix, 2023-12-25:
Hi,

please run `tbot`/`newbot` with an additional `-v` flag to show channel communication and then post the log here.  Keep in mind to censor hostname/username/passwords...

Without the log, it is very hard to guess where the problem lies.

One general hint:  I recommend using the `SSHConnector` over the `ParamikoConnector` unless you have a good reason to use the latter.  The `SSHConnector` usually has much less issues.

Adnan-Elhammoudi, 2023-12-25:
Hi,

the same issue has been applied with  LocalHost, a board, and a board's Linux environment:

code:

```
import tbot
import json
from tbot.machine import connector, board, linux
from tbot.machine.linux.linux_shell import LinuxShell

class Device(connector.ConsoleConnector, board.Board):
    with open('config/config.json', 'r') as file:
        data = json.load(file)
        baudrate = data['DUT']['baudrate']
        serial_port = data['DUT']['serial_port']
        user_name = data['DUT']['username']
        pass_word = data['DUT']['password']
        login__prompt = data['DUT']['login_prompt']
        password__prompt = data['DUT']['password_prompt']
        file.close()

    def connect(self, mach):

        ch = mach.open_channel("picocom", "--quiet", "-b", str(Device.baudrate), "-r", "-l", Device.serial_port)
        return ch

class DeviceLinux(board.Connector, board.LinuxBootLogin, linux.Bash):

    login_prompt = Device.login__prompt 
    password_prompt = Device.password__prompt
    username = Device.user_name
    password = Device.pass_word
    login_delay = 5
    boot_timeout = 10

def register_machines(ctx):
    ctx.register(Device, tbot.role.Board)
    ctx.register(DeviceLinux, tbot.role.BoardLinux)
```
logs:

```
==================================================================== test session starts =====================================================================
platform linux -- Python 3.8.10, pytest-7.3.1, pluggy-1.0.0 -- /usr/bin/python3
cachedir: .pytest_cache
metadata: {'Python': '3.8.10', 'Platform': 'Linux-5.15.0-91-generic-x86_64-with-glibc2.29', 'Packages': {'pytest': '7.3.1', 'pluggy': '1.0.0'}, 'Plugins': {'metadata': '3.0.0', 'html': '4.1.1'}}
rootdir: /home/adnanelhammoudi/Documents/system-test
plugins: metadata-3.0.0, html-4.1.1
collected 3 items                                                                                                                                            

test_cases/test_connectivity.py::test_hardware_platform[x86_64] │   ├─[local] picocom --quiet -b 115200 -r -l /dev/ttyUSB2
FAILED
test_cases/test_connectivity.py::test_hardware_platform[x86] │   ├─[local] picocom --quiet -b 115200 -r -l /dev/ttyUSB2
FAILED
test_cases/test_connectivity.py::test_hardware_compatibility │   ├─[local] picocom --quiet -b 115200 -r -l /dev/ttyUSB2
^C

========================================================================== FAILURES ==========================================================================
_______________________________________________________________ test_hardware_platform[x86_64] _______________________________________________________________

architecte = 'x86_64'

    @pytest.mark.parametrize("architecte",[HARDWARE_PLATFORMS[0],HARDWARE_PLATFORMS[1]])
    def test_hardware_platform(architecte):
    
>              with tbot.ctx.request(tbot.role.BoardLinux) as lnx:

test_cases/test_connectivity.py:12: 
[... 319 lines cut ...]

    def read(self, n: int, timeout: typing.Optional[float] = None) -> bytes:
        if not self.closed:
            # If the process is still running, wait for one byte or the timeout
            # to arrive.  We run select(2) in a loop to periodically (each
            # second) monitor whether the subprocess is still running.
    
            end_time = None if timeout is None else time.monotonic() + timeout
            while True:
                if end_time is None:
                    select_timeout = MIN_READ_WAIT
                else:
                    select_timeout = min(MIN_READ_WAIT, end_time - time.monotonic())
                    if select_timeout <= 0:
>                       raise TimeoutError()
E                       TimeoutError

/usr/local/lib/python3.8/dist-packages/tbot/machine/channel/subprocess.py:81: TimeoutError
---------------------- Generated html report: file:///home/adnanelhammoudi/Documents/system-test/report.html ----------------------
================================================================== short test summary info ===================================================================
FAILED test_cases/test_connectivity.py::test_hardware_platform[x86_64] - TimeoutError
FAILED test_cases/test_connectivity.py::test_hardware_platform[x86] - TimeoutError
FAILED test_cases/test_connectivity.py::test_hardware_compatibility - TimeoutError
===================================================================== 3 failed in 31.11s =====================================================================
```

Adnan-Elhammoudi, 2023-12-29:
it works when I have changed the `login_prompt = " "` to white space , I don't know why !

Rahix, 2023-12-29:
Hm, `login_prompt = ""` looks very wrong.  However, please try

```
login_prompt = "~# "
#                 ^- notice the space here
```

Adnan-Elhammoudi, 2024-01-08:
I have tried with and without a space `"~# " "~#" `, the issue here is the login prompt does not show up when reading the stdout until pressing ENTER,

it hangs on the last log message from the kernel

[image]

Rahix, 2024-01-11:
Please provide some more context.  Are you sure there isn't a login prompt somewhere between the kernel log messages during boot?  Can you share a bit more of the bootlog maybe?

Adnan-Elhammoudi, 2024-01-12:
it has automatic login actually and run a script, but i was trying to catch the console prompt to run the test using software abstract role.
and I can't detected it until pressing enter, even manually I have to do so.
I tried to send enter as control character over the channel but it does not works.

closed 2024-02-26 by Adnan-Elhammoudi