Rahix/tbot #110: Send arbitrary commands via uboot console

Adnan-Elhammoudi, 2024-01-25:
Is there any way to send the Enter key as an arbitrary command via u-boot console?
My purpose here is to make the `md` increment the next result address.

Rahix, 2024-01-26:
I would avoid doing that.  This "enter without a command does something special" behavior of U-Boot is an optimization for allowing CLI users to type less. But tbot doesn't care about how much "typing" it has to do.  So it is a better idea to send explicit commands every time rather than relying on such convenience features on the other side.  This also gives you a lot more flexibility should you ever want to adjust your code.

So I'd rather do something like this:

```python
START_ADDR = 0x10000000
READ_SIZE = 0x100

addr = START_ADDR
for i in range(10):
    output = ub.exec0("md.b", hex(addr), hex(READ_SIZE))
    addr += READ_SIZE
```

Rahix, 2024-02-15:
@Adnan-Elhammoudi is this topic resolved for you?  If yes, please close the issue :)

closed 2024-02-16 by Adnan-Elhammoudi