Rahix/tbot #122: Handle multiple serial connections

DiogoSilva014, 2024-11-01:
Hi!

I have a use-case where I need to monitor two serial outputs (for logging purposes), although I only need to interact with one of them. The kernel is outputting to one serial (which I need to log for troubleshooting failed tests, but not to parse or anything similar), and the second serial output should be used to interact with the board. I tried having a thread only dedicated to logging the kernel serial but sometimes it happens that both serials output in the middle of a line and the output becomes gibberish.

Is there any straightforward way to do this? 

Thanks and best regards

Rahix, 2024-11-01:
Hi,

don't ever try to use tbot with multithreading, this will go wrong in the worst of ways.  tbot is meant for strictly serial interaction with the system under test.

The way I usually work with these kinds of situations is to interleave reads/interactions of both consoles in an appropriate way.  The specifics always depend on the concrete testcase, but here is an example:

```python
with tbot.ctx() as cx:
    log_console = cx.request(config.role.LogConsoleBoard)
    linux = cx.request(tbot.role.BoardLinux)

    ev = cx.enter_context(tbot.log.EventIO(
        ["board", "log"], "", verbosity=tbot.log.Verbosity.STDOUT
    ))
    # A different prefix so messages can be distinguished
    ev.prefix = "   <> "
    cx.enter_context(log_console.ch.with_stream(ev))

    # Read log for 2 seconds
    log_console.ch.read_until_timeout(2)

    linux.exec0("uname", "-a")

    # Now that the uname has finished, read the log again (without any delay)
    log_console.ch.read_until_timeout(None)

    with linux.run("./long-running-tool") as long_running_tool:
        try:
            while True:
                # We want to read both consoles semi-simulataneously until the command finishes
                long_running_tool.read_until_timeout(3)
                log_console.ch.read_until_timeout(None)
        except linux.CommandEndedException:
            long_running_tool.terminate0()
```

Hope this helps!

DiogoSilva014, 2024-11-01:
Hi @Rahix ,

Yes, that does indeed help. Thanks!

closed 2024-11-01 by DiogoSilva014